rusefi/hellen128merc-issues #10: handle supecharger clutch f21

duhafnusa4, 2021-07-17:
On Hellen 128 Pin Mercedes Pin KF21 is associated to ignition, but on the stock ECU that pin is for engaging the supercharger clutch (pin pulls low to trigger relay that feeds clutch 12V)

Need to move igntion pin

rusefillc, 2021-07-17:
So @duhafnusa4 has https://github.com/rusefi/rusefi/wiki/Mercedes-C230-and-SLK230

rusefillc, 2021-07-17:
while v8 has https://github.com/rusefi/rusefi/wiki/1999-e430

mck1117, 2021-07-17:
Maybe not move, but jumper. Some cars have ign on that pin.

mck1117, 2021-08-16:
[image]

rusefillc, 2021-08-18:
@mck1117 how did you hack this board for now?

mck1117, 2021-08-18:
I didn't really hack the board, @duhafnusa4 hacked his harness.

He moved ign 1 is to one of the pins on channel5 (F4 or F5).

His supercharger clutch has a relay, connected to pin F21, so the IGBT is installed there as a low side driver.  This is NOT the factor wiring for the supercharger.

rusefillc, 2021-08-18:
Matthew Kennedy  5:18 PM
Huh?

andrey  5:18 PM
@Jimmy soiboi Luntzel says "relay" in ticket and "low"

Matthew Kennedy  5:18 PM
Yes
5:18
He added a relay so the ecu switches lowside
5:18
Because he had MRE before
5:18
IGBT can switch lowside

andrey  5:18 PM
https://github.com/rusefi/rusefi/wiki/OEM-Docs/Mercedes/99_SLK230_3.png
5:18
does not OEM switched lowside?

Matthew Kennedy  5:18 PM
Dunno
5:18
Where does that wire go left side?

andrey  5:19 PM
see https://github.com/rusefi/rusefi/wiki/OEM-Docs/Mercedes/99_SLK230_3.png

Matthew Kennedy  5:19 PM
We thought it was maybe high side but I never actually looked at the schematic

andrey  5:19 PM
also https://github.com/rusefi/rusefi/wiki/OEM-Docs/Mercedes/99_SLK230_2.png

Matthew Kennedy  5:19 PM
Looks low side to me

andrey  5:19 PM
not ECU wire goes to +12v

Matthew Kennedy  5:19 PM
Yeah

andrey  5:19 PM
but also high current

Matthew Kennedy  5:20 PM
@Jimmy soiboi Luntzel you around? Can you measure ohms of the blower clutch?

andrey  5:20 PM
maybe @Jimmy soiboi Luntzel needed high current so he added relay since MRE not high current?

Matthew Kennedy  5:20 PM
Igbt can do huge current but they're inefficient
5:20
*at low voltages

andrey  5:20 PM
how is Igbt not what board has for igniters?
5:20
i.e. is "Igbt" different from what we use for dump coils?

Matthew Kennedy  5:21 PM
Dumb coils have exactly igbt

andrey  5:21 PM
or is the point that Jimmy needed 3rd Igbt or first Igbt due to smart coils?

Matthew Kennedy  5:21 PM
If that clutch pulls 5 amps, it's too much heat

5:21
He has 4 jumpers for smart coils on channels 2345

andrey  5:21 PM
"jimmy added Igbt like rev A is asking for for dump coil"

Matthew Kennedy  5:21 PM
And an igbt acting as lowside switch on SC
5:21
On ign1

andrey  5:22 PM
so a better hack would be through hole mosfet soldered to those pads
5:22
or just a mosfet with matching pinout maybe?

Matthew Kennedy  5:22 PM
I think most/all mosfets have backwards pinout from IGBT
New
5:23
That doesn't solve the problem that ign output 1 and blower are tied together on those two pins

andrey  5:24 PM
what pins? why tied together?
5:24
what car seems problematic?

rusefillc, 2021-08-18:
Matthew Kennedy  
both F20 and F21 are IGN1 on some car

I don't know if any car uses F20=IGN and F21=SC at the same time

rusefillc, 2021-08-19:
Matthew Kennedy  5:36 PM
so yes, it needs either a relay or a bodgewire
5:37
*bodge mosfet

Qwerty-OFF, 2021-08-22:
Can we use any unused IGBT for this?
[image]

rusefillc, 2021-08-22:
I think there is heat dissipation concerns? Do you mean multiple IGBTs in parallel? Maybe 4 IGBTs in parallel is the answer @mck1117 ?

mck1117, 2021-08-23:
noo, please don't do that.  IGBTs have a roughly fixed voltage drop (not resistance), and it's like ~2 volts for these IGBTs.  Blower clutch could pull north of 4-5 amps.

Qwerty-OFF, 2021-08-23:
Can we use BTS3028 or something else in d2pak package for replace this IGBT?

rusefillc, 2021-08-24:
we are not REPLACING right, we are ADDING an alternative option for when we need clutch control not spark coil control?

rusefillc, 2021-08-24:
BTS3028 

[image]

rusefillc, 2021-08-24:
ISL9V3040S3S
[image]

rusefillc, 2021-08-24:
@Qwerty-OFF please share exact full PN of BTS3028 with proper package

andreika-git, 2021-08-24:
pinout looks good

mck1117, 2021-08-24:
it might be a little marginal on current capacity, the clutch could be 5 amps, which on a bts3028 would be 2.3 watts.  Anything available with lower R_ds_on?

Besides, the BTS3028 is DPAK, not D2PAK.

Qwerty-OFF, 2021-08-24:
VNB14NV04
BTS3018TC
BTS141TC
BTS133TC

mck1117, 2021-08-24:
I like that ST part if availability is ok

Qwerty-OFF, 2021-08-24:
[image]