rusefi/hellen128merc-issues #13: analog inputs (limitation)

shadowm60, 2021-08-16:
From the schematic point of view there are more adc channels then exposed to CAR connector.

All unused channels should have a Test Point, this way user can extend functions, ex. add fuel pressure sensor, AC compressor, oil pressure, etc

rusefillc, 2021-08-16:
@shadowm60 can you please be more specific what exact STM32 pin seems unused and not routable?

mck1117, 2021-08-16:
[image]

Some of these have vias, but most are no-connect, why?

shadowm60, 2021-08-22:
We had a discussion with @mck1117 and we cannot use all the pins as analog since we do have a limitation for all those unconnected channels, but then i was thinking, can we use them at least like inputs? ex. ac compressor on, clutch pedal, starter switch, etc?

Qwerty-OFF, 2021-08-22:
We have very little space on the board for test points. I can try adding a few more, but definitely not all. What exactly do I need to connect to test points?

shadowm60, 2021-08-22:
Why not loose the WBO module two conenctor setup ( i mean if we have the test points for soldering in a wbo connector, why do we have the plastic connector in place as well?) user would use only one, probably that would not free up space on the needed area, but still as a user i see no point into having two sets of connectors for wbo. i would be much happier if i can extend my setup with additional inputs

rusefillc, 2021-08-22:
@Qwerty-OFF we need to find a way. Maybe #16

Qwerty-OFF, 2021-08-23:
Added three analog inputs
[image]