rusefi/hellen128merc-issues #14: ignition fuse wrong value

shadowm60, 2021-08-18:
Using oem coils and IGBT blown the fuses on ignition output, wrong value selected.
IGBT would not get how so probably value should be something close to the limit of igbt?

mck1117, 2021-08-18:
we can go for a much higher value

current fuse is 5A rated

IGBT is rated for ~20 amps continuous, but that would dissipate ~40 watts, so we're thermally limited much lower than that (dc), so maybe 10-20 amps is a better fuse rating

Qwerty-OFF, 2021-08-22:
We need a jumper that:
1. Withstands 400-500V
2. Does not burn out at a current of 10-20A
3. Sold and assembled at JLC
4. Takes up little space
Have any ideas? If we refuse to JLC accembly, we can use a wire.

rusefillc, 2021-08-22:
I am totally not in the loop, why do we need fuses/wires at all? is this about optional gap jumping or else?

Qwerty-OFF, 2021-08-22:
This is a replacement for our 0805 smd jumper in case we want to manually change pinout. It is impossible to use 0805 smd due to high current and voltage

rusefillc, 2021-08-22:
Realistically it looks like jumpers we do not need. Can we have traces with vias? So normal case traces are used, funny wiring which is not probable we cut traces with a knife and use holes for wires?

Qwerty-OFF, 2021-08-22:
Looks like we need a jumper at least here https://github.com/rusefi/hellen128merc-issues/issues/10
The idea with a knife can be realized

mck1117, 2021-08-23:
Ehh, it's still a good idea to have the fuse, as we've seen.  Alex found a firmware bug that probably would have melted his coils, but was saved by the fuse.

Unclear if he blew the fuse in normal operation though, and not during a true fault.

Does JLC not have any >5 amp fuses?

Qwerty-OFF, 2021-08-23:
Maybe something like 0505016.MXEP C187650, but I'm not sure that using a ceramic fuse in car is a good idea.

The fuse already used is very slow, I was hoping that it would not blow out from the impulse current. Are there other users of the board? Didn't they have such a problem?
[image]

rusefillc, 2021-08-24:
[image]

rusefillc, 2021-08-24:
[image]

Qwerty-OFF, 2021-08-24:
We don't have space for fuses like https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C187650
Let's try using 250V 10A fuses https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C354908