rusefi/hellen128merc-issues #16: Space saving measures

rusefillc, 2021-08-22:
I agree that we can loose second wire soldering option of WBO

We can also move battery to the back side of the board with hand-soldering if desired

rusefillc, 2021-08-22:
smart coil resistors like R78 can also go on the back since not populated by default

rusefillc, 2021-08-22:
@andreika-git @mck1117 any other ideas?

one day we will get dual side assembly

mck1117, 2021-08-23:
how much space do we need?

switch to two pin header for battery? there are batteries with a short two wire pigtail available off the shelf

mck1117, 2021-08-23:
> smart coil resistors like R78 can also go on the back since not populated by default

I wouldn't move those, since we're already constrained by the "case wall" there, so that stuff all has to be on the far side of it