rusefi/hellen128merc-issues #23: Document aux input hardware options

rusefillc, 2021-09-11:
how can we feed 2nd Manifold air pressure sensor. So 0-5v analog signal into ECU?

shadowm60, 2022-01-15:
I have started an xls on this topic do have all the inpus documented with there config (pullup and pull down)

rusefillc, 2022-01-15:
We have too many parallel processes, lets agree on a more common process

shadowm60, 2022-01-15:
I did not know of a process, is there a way for me to gelp on this ticket?

abelom, 2022-01-15:
We have pinout yaml maybe there

Some boards have table

Between pinout yaml, ineractive pinout and wiki pages we definitely need
more copies of same imho, we need to get some copy useable

rusefillc, 2022-01-15:
I will document what we have better

shadowm60, 2022-01-15:
My point was, yes yaml is ok that would connect user on ts side to use it, but we need a littel more like is it a pull.up input, that way a switch to gnd can be placed there... i am trying to think as an end customer who is doing a list what can be done additional to oem harness

rusefillc, 2022-01-15:
@shadowm60 I need help adding all useful "Pxx" aux points into merc yaml file

Specific example https://github.com/rusefi/rusefi/commit/7ee9c76754391a7af5a3dbc9708d3d91b96a0772

what makes this ask tricky is the part that many hellen inputs/outputs are not really documented so far, with module source code being the only documentation

shadowm60, 2022-01-27:
yaml file was updated, probably this needs to be taken care of at every hw update. Right now i feel pretty good, wiki and interactive pinout is somehow connected.

closed 2022-01-27 by rusefillc