rusefi/hellen128merc-issues #44: E3 wastegate low or high? also diode

rusefillc, 2021-11-01:
also document diode mod for older revisions

rusefillc, 2021-11-01:
[image]

rusefillc, 2021-11-01:
@mck1117 @Qwerty-OFF @andreika-git 

does this image of the weird three wire solenoid (?) actuator (?) tell us if it's low-side or high-side driven? is that logic-level control or high current control? what can we tell about this recirculated air flap actuator from this schematics?

[image]

Qwerty-OFF, 2021-11-01:
It looks like a transistor inside. Possibly controlled by PWM logic level.

Qwerty-OFF, 2021-11-01:
For rev. b:
Remove R10
[image]
Wire from J11 (left contact) to R10 (upper contact)
[image]
You can try use SPI3_MOIS (PC12) as soft-pwm

rusefillc, 2021-11-01:
That's assuming transistor would be happy with 3v

Maybe better use unused ign output? Surface pad soldering of wire but 5v?

Qwerty-OFF, 2021-11-01:
For 5v levels, you can use IGN5 output (PE2)
[image]

rusefillc, 2021-11-06:
```
we tried to get the wastegate running with the 5v pmw signal but it doesnt react. We think the internal electronics might expect 12V amplitudes of the pwm. Is there a possibility to get a 12v pwm Signal running from the hellen```

rusefillc, 2021-11-06:
```
Not with the osci but we measured a peak of 12v on the signal input with original ecu when wastegate is operated to maximum
```

Qwerty-OFF, 2021-11-06:
Try use any low-side (P19, P34, P35) with external 12v pullup resistor (4.7k for example)

rusefillc, 2021-11-06:
@Qwerty-OFF is there a place on the PCB where we have an unused 4.7k to source? :)

Qwerty-OFF, 2021-11-06:
R51, R52 used for baro sensor may be moved to proto area and connected to 12v