rusefi/hellen128merc-issues #49: Aux H-bridge channel

rusefillc, 2022-01-12:
Alex is looking to control aftermarket DC motor wastegate. Let's split second channel of U1 out of ETB duty

[image]

rusefillc, 2022-01-12:
looks like we have some unused MCU pins?

[image]

shadowm60, 2022-01-15:
I took a look in tge datasheet of the h-bridge. Current config good for 5.6A. Single channel can do half of that, when etb is stuck and we try to move it we consume more then 3A. Can we do some protection? Do we relay only on driver for protection? Miss config ow wireing could blow this?

rusefillc, 2022-01-15:
@shadowm60 driver has protection. We rely on driver. We can also look into alternative part numbers altogether.

rusefillc, 2022-01-15:
I have stock of TLE9201 we can go two units of TLE9201.
We can also go with same TLE9202 as Proteus is using.

Open question how painful is to hand solder TLE9202 vs two of TLE9201 vs L6205 @mck1117 @OrchardPerformance

shadowm60, 2022-01-15:
I played with mre 0.3 and kompressor bypass solenoid, and i noticed peeks over 3A. So if we could implement two channels with current limits it would be nice.

rusefillc, 2022-01-15:
@Qwerty-OFF sounds like NOT split channels on current chip

We can keep ETB as is and ADD tle9201, or replace.

shadowm60, 2022-01-15:
Where to output the new motor outputs.

This is hard to answer, mainly this is one point i did nit opened the ticket myself. I was thinking of using wires to repurpose some ignition output pins. Most common setup are l4 and l6 so we do not need all the outputs.
When i get home i can look up a specific pair, but i do nkt know how you feel about soldering wires between test points

shadowm60, 2022-01-15:
Two simmetrical unit would make more sense? I mean why build full spec, only on customer request populate the second channel? We are doing wbo the same way... simple to handle chips this way and repair jobs.

rusefillc, 2022-01-19:
for aux H-bridge let's have pads same as on V12 BMW adapter board same Molex 2 pin?

shadowm60, 2022-01-20:
With pads to solder i agree, molex pin? I do not know what you mean by that, i think if it is used, we can remove unused ignition output fuse, and solder hbridge to the desired bin, car harness repinned and you have new function available

shadowm60, 2022-01-20:
So i have checked my car wiring, it uses KF20 as igniter, on hellen f20 and f21 are both OUT_IGN1 and there is a fuse in the way, so for me it looks manageable, to remove the fuse for ex on KF21 and solder a wire for h-bridge.

rusefillc, 2022-01-20:
@shadowm60 is that original wiring on your vehicle? do we have your schematics on our wiki? See also https://github.com/rusefi/rusefi_documentation/issues/197

shadowm60, 2022-01-20:
Mine is original, coils use only one pin, i was browsing today to find out why do we use two pins, but i know that m112 has two coils/cilinder, that would explain, but right now i do not have the big picture

shadowm60, 2022-01-21:
My schematic is not on wiki, i found the xls where i have entered my connection, i can upload the schematic for Vault of mercedes if it is needed https://docs.google.com/spreadsheets/d/1I-lZKRajTiEGFUXdZpXEtKF2pymlOo-lPahy3cLMnl4/edit#gid=489142936 in the column with C180 Alex use (M111.921 MY99)

shadowm60, 2022-01-21:
https://github.com/rusefi/rusefi/wiki/Vault-Of-Mercedes-OEM >> OEM Wiring harness
Or direct link: https://github.com/rusefi/rusefi/wiki/OEM-Docs/Mercedes/w202c180m111wiring4.pdf
I have also added the ME2.1 description that i found: https://github.com/rusefi/rusefi/wiki/OEM-Docs/Mercedes/motronic.pdf

rusefillc, 2022-01-21:
@shadowm60 now that we have a link just to reduce confusion are there any specific pins we need to focus on for the purposes of this "aux DC" ticker?

shadowm60, 2022-01-21:
Relevant is that for I4 m111 engine we do not have only 2 coils, and are driven only from 2 pins. Every ignition pin is connected via a fuse to the car connector, so if we go with wires, we can remove unused fuse and use it as additional DC motor. So i do not have a specific pin that we could route next to. All i can say that F13 and F20 is used in the factory harness, So at the end it all depends if the H-bridge output are close to F4, F5 i would probably say that is what we can use.

shadowm60, 2022-01-21:
Let me try it again. I cannot chose 2 pins. We know we would need wires. Simple approache, remove two fuses and solder the wire to the car connector end of the fuse footprint. If the layout would make it possible to do the same, only no wires, but jumpers, then i think layout would chose the two pins, and we avoid wires.

I do not have pins that i need/would use, my wiring diagram just highlighted the it only uses 2 pins for ignition, the rest are free.

closed 2022-01-26 by rusefillc