rusefi/hellen128merc-issues #50: Vertical USB type B

rusefillc, 2022-01-25:
does C26234 look right?

Jing Extension of the Electronic Co.
921-111A1010Y10200

rusefillc, 2022-01-25:
TE part

[image]

rusefillc, 2022-01-26:
done on rev C

abelom, 2022-01-26:
```
Hi Sir or madam,

Sorry for one more question.

2) It seems that the J8 is not match with the through holes on PCB, so you may failed to install it.

Shall we skip it and refund the cost? Or enable the replace file button?

Or do you prefer to take the risk and continue make it?

Your early reply will be highly appreciated, thank you so much!
```

[image]

Qwerty-OFF, 2022-01-26:
[image]
[image]

abelom, 2022-01-26:
[image]