rusefi/hellen128merc-issues #51: JLC says slots

rusefillc, 2022-01-26:
is that on ign8 module on or frame @Qwerty-OFF @andreika-git 

[image]

rusefillc, 2022-01-26:
plated slots they say?

[image]

Qwerty-OFF, 2022-01-26:
Frame:
[image]
Board:
[image]
This is a pair of holes with similar coordinates
@andreika-git any ideas?